NVIDIA-AI-IOT/jetracer #106: Power supply to the Servo Driver

tran9774, 2021-06-12:
Hi guys.

I'm following this build but I didn't purchase a rc controller because I wanted to use my xbox controller for it.
How do you supply power to the PCA9685 to move the Servo?

Greetings from Germany.

akwong922, 2021-06-14:
I did something similar with a PS3 controller. The PCA9685 powers up when connected to the ESC when powered on and i can control the throttle and the steering from it. Not sure if this will cause any problems in the long run though.

tran9774, 2021-06-14:
How did you setup the ESC in that case. Because i don't have a radio transmitter the ESC keeps beeping.
I can control the steering with the Adafruit servo kit library in Python though.

wackerl91, 2021-07-08:
I have pretty much the same setup. Here is what I did:
- Jetson Nano is connected to the PCA9685 using the usual 4-wires connection (powers on with the Jetson)
- Steering servo is connected with all three pins on the PWM headers
- ESC is connected to the PCA9685 using only the PWM signal on the PWM header (not GND/V+, so the Jeston only sends the PWM signal via the PCA9685 board)
- GND/V+ to PCA9685 is supplied from ESC using the green/blue power terminal connector (opposite of the PWM header on the PCA9685 board; not sure if this is strictly necessary as I haven't tried disconnecting those cables)

To my understanding (not an electrics guy ...), this way the servos draw power via the ESC from the car battery instead of from the Jetson - on Adafruit there's a warning that excessive current may cause resets and/or damage when an Arduino provides the power, so I figured the same may be the case for the Jetson.

Regarding the ESC, it beeps as long as no control input has been received. I set both steering servo and motor to 0 when launching my script before switching on the ESC.

Feel free to shoot me a mail if some photos would help you with wiring everything up.

Cheers, and have fun with your Jetracer! :)

dbanshee, 2021-11-18:
Hi guys, building a similar configuration too, I've noticed (at least in my setup) that Voltage on PWM cables from ESC are higher than reasonable 5V. I read 5.9V and this is propagated to PCA9685 and Jetson (really an rpi, but it is not rellevant). So I think this could be dangerous.

I'm not using a battery. I'm using an ATX PSU instead, converting from 12V to 7V with DC-DC stepper down.
Can you check the theorically 5V ESC line with full battery?

I've created a new issue with more details before you are interested.

Grettings from Spain.

closed 2022-07-18 by jaybdub