NVIDIA-AI-IOT/jetracer #111: road_following.ipynb import torch OSError

AntonioCervantes, 2021-07-14:
Hello, I am using the Nvidia Jetson Nano 4gb B01 with the latest Jetracer SD card image (4.5.1) and all the same electronic hardware from the Bill of Materials for the Tamiya TT02 RC car. After following all the software setup from the Jetracer GitHub page, I was able to run through the basic_motion and interactive_regression notebooks just fine with no issues. However after getting to the road_following notebook and running the first block of code I kept getting an error that says: `OSError: /usr/lib/aarch64-linux-gnu/libgomp.so.1: cannot allocate memory in static TLS block`

I was able to get this notebook to work once by running some of the code out of order, but my CSI camera couldn't connect and I had to restart. I wasn't able to replicate this out-of-order code execution and have been stuck on the `import torch` line of code ever since. Could someone please help me out with this issue or share any info on how to solve it, thanks!

[image]

diamondbarcode, 2021-11-08:
i try multiple answers that i found via google but cant seem to fix this problems .. if anyone got solution to this please write guides ..

crother94, 2021-12-22:
I had the same problem. I saw that several of the solutions from google mentioned switching the order of importing torch and cv2. Since this notebook didn't include cv2, I added a line "import cv2" above the other two imports and it fixed the error. I'm not sure why it worked, or if it's the best way of getting past the error, but hopefully it helps you.