NVIDIA-AI-IOT/jetracer #112: Dynamically adjusting JetRacer speed

HamburgDave, 2021-07-24:
Has anyone tried to change JetRacer's speed (throttle) based on the images or maybe y-axis values?  For example having a higher throttle on a straight section, slower in turns?

arickchan, 2021-09-24:
I have using the floting value x , and change it to absolute value [ abs (x) ], use a formula 1 - abs (x), if x is at center (value = 0), the car can run with original setting speed, if x has +/- value, the speed will be reduced.

arickchan, 2021-09-24:
But I think can have a better method to setup better dynamic speed of the car

HamburgDave, 2021-09-27:
Thank you for the reply and this sounds quite interesting!   I'd be interested in learning more if you're available.  

There may be a demonstration of scale (1:10/1:5) cars in conjunction with the Indy Autonomous Challenge event on Oct. 23.  It would be great, if possible, to incorporate your approach into the demo.   Let me know if you're interested and we can discuss it further.  

The other challenge I've been facing is training JetRacer to execute an overtaking (passing) maneuver.  I think it can be done without adding a lidar but not sure.  Would be interested in your thoughts on this as well if it's something you've considered.  Thanks again for the reply and looking forward to connecting.