NVIDIA-AI-IOT/jetracer #113: interactive regression error

songggggggg, 2021-08-31:
import torchvision.transforms as transforms
from xy_dataset import XYDataset

TASK = 'road_following'

CATEGORIES = ['apex']

DATASETS = ['A', 'B']

TRANSFORMS = transforms.Compose([
    transforms.ColorJitter(0.2, 0.2, 0.2, 0.2),
    transforms.Resize((224, 224)),
    transforms.ToTensor(),
    transforms.Normalize([0.485, 0.456, 0.406], [0.229, 0.224, 0.225])
])

datasets = {}
for name in DATASETS:
    datasets[name] = XYDataset(TASK + '_' + name, CATEGORIES, TRANSFORMS, random_hflip=True)

---------------------------------------------------------------------------
OSError                                   Traceback (most recent call last)
<ipython-input-5-b02138784043> in <module>
----> 1 import torchvision.transforms as transforms
      2 from xy_dataset import XYDataset
      3 
      4 TASK = 'road_following'
      5 

~/.local/lib/python3.6/site-packages/torchvision/__init__.py in <module>
      4 from .extension import _HAS_OPS
      5 
----> 6 from torchvision import models
      7 from torchvision import datasets
      8 from torchvision import ops

~/.local/lib/python3.6/site-packages/torchvision/models/__init__.py in <module>
----> 1 from .alexnet import *
      2 from .resnet import *
      3 from .vgg import *
      4 from .squeezenet import *
      5 from .inception import *

~/.local/lib/python3.6/site-packages/torchvision/models/alexnet.py in <module>
----> 1 import torch
      2 import torch.nn as nn
      3 from .utils import load_state_dict_from_url
      4 from typing import Any
      5 

~/.local/lib/python3.6/site-packages/torch/__init__.py in <module>
    194     # See Note [Global dependencies]
    195     if USE_GLOBAL_DEPS:
--> 196         _load_global_deps()
    197     from torch._C import *  # noqa: F403
    198 

~/.local/lib/python3.6/site-packages/torch/__init__.py in _load_global_deps()
    147     lib_path = os.path.join(os.path.dirname(here), 'lib', lib_name)
    148 
--> 149     ctypes.CDLL(lib_path, mode=ctypes.RTLD_GLOBAL)
    150 
    151 

/usr/lib/python3.6/ctypes/__init__.py in __init__(self, name, mode, handle, use_errno, use_last_error)
    346 
    347         if handle is None:
--> 348             self._handle = _dlopen(self._name, mode)
    349         else:
    350             self._handle = handle

OSError: /home/jetson/.local/lib/python3.6/site-packages/torch/lib/libgomp-d22c30c5.so.1: cannot allocate memory in static TLS block

i don't know why this happen.
plz helppp

diamondbarcode, 2021-11-08:
I had this problem too cant find the proper solution to it yet. Any luck?

StayAlone01, 2021-12-13:
Just refresh the website, if it still doesn't work, restart the kernel.