NVIDIA-AI-IOT/jetracer #117: Jetracer 2GB expansion board

yahya200, 2021-11-16:
I have purchased jeracer 2GB Al kit .
I assembled the hardware by following the instructions in Waveshare wiki , but unfortunately it doesn’t power up and OLED doesn’t display anything when I turn it on. Jetson nano is working properly getting the power from expansion 
[image]
board. What is the problem?please help 

Thanks

yongen9696, 2022-05-08:
Hi, I would like to know whether you have solved the issue? My version is a 4GB Pro kit, has the same problem as yours where the OLED display doe not lights up.

K-Molloy, 2022-05-09:
I also have this problem with the 2GB kit, no OLED or motor - but steering is okay

yongen9696, 2022-05-31:
Updates on mine:

1. I have contacted the WaveShare support team, turns out is my expansion board is broken.
2. Bought a new one from their sales team directly, cost me 46.99 USD for the **expansion board only**.

alternative, if you have a device like Arduino, you can DIY with the Jetson nano.

yahya200, 2022-10-11:
https://jetsonhacks.com/2019/12/03/adafruit-pioled-on-jetson-nano/
Hi ,follow steps and enjoy