NVIDIA-AI-IOT/jetracer #123: How to use with 2GB jetson nano ? Instructions for building/packaging JetCard image ?

c1505, 2022-01-06:
I have a 2GB jetson nano.  It seems like it is useable based on an issue I found https://github.com/NVIDIA-AI-IOT/jetracer/issues/100  and that this company sells a 2GB kit https://www.waveshare.com/jetracer-2gb-ai-kit.htm?sku=20557 .  In the mentioned issue, it is stated that you can make it work by "preparing the card by hand".  I don't see any instructions for how this software is built and packaged so I am not not sure what preparing it by hand would mean.  If someone could help me with that, I would appreciate it.  I did find this link from [waveshare](https://drive.google.com/file/d/1YtnjQ77w1B9REzy1JgLJbVSs2K3ocAEr/view) with an image file for the 2GB version.  I would prefer to be able to create the image myself or have it from a more known source like NVIDIA.   

Maybe @TigroTigro or @waveshare could help add some clarity here.  Thanks.

yoshiorubio, 2022-01-14:
You can download the 2GB image for the Jetson nano following the instructions:
<https://developer.nvidia.com/embedded/learn/get-started-jetson-nano-devkit>

Then proceed to follow the software installation instructions:
<https://github.com/NVIDIA-AI-IOT/jetracer/blob/master/docs/software_setup.md>

You would need to install Torch for the Jetson nano to run any deep learning algorithm, so just follow:
<https://forums.developer.nvidia.com/t/pytorch-for-jetson-version-1-10-now-available/72048>

c1505, 2022-01-14:
Ah ok.  Thanks.  So the jetcard image is the jetpack/nano-devkit v4.5.1 + pytorch + jetcam + torch2rt + jetracer ? 
Do you know the release or commit that is installed for each of those packages ?

c1505, 2022-01-26:
What worked for me so far was using the JetBot pre-built SD card image for JetPack version 4.5 and then installing additional software on top of that.  Because of the similarities between this project and the JetBot project, that seems like the fastest and easiest approach.  I am able to run through the notebooks I have tried without error .  I don't have the RC car hardware yet so I can't guarantee this approach will work then, but it seems likely. 

Before that, I tried starting from the Jetpack version I had installed, upgraded it, and tried to install software on top of it.  I partially went through that.  Because it was time consuming and I wasn't sure what versions of software I should be using, I switched to starting from the JetBot.  Starting just from Jetpack, at minimum the additional software needed as compared to JetBot is PyTorch, torchvision,  and jupyterlab .  There is also configuration done to disable the display for JetBot so that the Nano isn't using compute resources on the display.  This seems even more important for the 2GB version of the Nano.       

### Instructions to set up Jetracer on a 2GB Jetson Nano

* [ Follow software setup Using SD Card Image for JetBot](https://jetbot.org/master/software_setup/sd_card.html) for the 2GB Nano 
* Install setup tools `sudo apt-get install python3-setuptools`
* Install jetcam
```
git clone https://github.com/NVIDIA-AI-IOT/jetcam
cd jetcam
sudo python3 setup.py install
```
* Install jetracer software
```
git clone https://github.com/NVIDIA-AI-IOT/jetracer
cd jetracer
sudo python3 setup.py install
```
* Reboot the jetson nano

c1505, 2022-02-09:
Created a PR to add install instructions for 2GB nano.  https://github.com/NVIDIA-AI-IOT/jetracer/pull/125

kn1ghtf1re, 2022-02-09:
@c1505 were you able to get the oled screen on the wavshare jetracer working?

c1505, 2022-02-09:
@kn1ghtf1re I don't have a waveshare .  Sorry that was unclear above.  I was just referencing them because it is the only 2GB image for jetracer I found.  They have their own fork of this code and images that I assume are slightly different.  If that is what you are working with, I would recommend contacting them .  This code repo doesn't have support for an oled screen as far as I can tell.

kn1ghtf1re, 2022-02-10:
Thanks @c1505. I'll check with them then!