NVIDIA-AI-IOT/jetracer #127: Trying to create a camera class- getting error message

MaxKru, 2022-03-09:
(https://forums.developer.nvidia.com/t/trying-to-create-a-camera-class-and-getting-this-error-message-could-not-read-image-from-camera/205455/2#here-is-my-full-error-output-1)

hello,

im trying to get get a camera working on my jetson nano, waveshare jetracer pro
using the B02 board,

heres my code:
#creating camera class

from jetcam.csi_camera import CSICamera
camera = CSICamera(width=224, height=224, capture_fps=65)

here is my full error output:

RuntimeError Traceback (most recent call last)
/usr/local/lib/python3.6/dist-packages/jetcam-0.0.0-py3.6.egg/jetcam/csi_camera.py in init(self, *args, **kwargs)
23 if not re:
—> 24 raise RuntimeError(‘Could not read image from camera.’)
25 except:

RuntimeError: Could not read image from camera.

During handling of the above exception, another exception occurred:

RuntimeError Traceback (most recent call last)
in 
1 from jetcam.csi_camera import CSICamera
2
----> 3 camera = CSICamera(width=224, height=224, capture_fps=65)

/usr/local/lib/python3.6/dist-packages/jetcam-0.0.0-py3.6.egg/jetcam/csi_camera.py in init(self, *args, **kwargs)
25 except:
26 raise RuntimeError(
—> 27 ‘Could not initialize camera. Please see error trace.’)
28
29 atexit.register(self.cap.release)

RuntimeError: Could not initialize camera. Please see error trace.

anybody got an idea how to fix this issue?

hs2415, 2022-08-19:
Hi,
 I'm having the same error in a different project (DLI Getting Started with AI Course) and haven't solved it with the methods I find on NVIDIA. Did you solve it in your case? 

Out of curiosity, do you have "jetcam-0.0.0-py3.6.egg/jetcam/csi_camera.py" located in /usr/local/lib/python3.6/dist-packages/? I don't and I wonder if it's related to the error. Hope I'm allowed to ask here!

MaxKru, 2022-08-22:
there is a camera reset service that should do the trick but didnt work for me,
https://github.com/NVIDIA-AI-IOT/jetracer/blob/master/notebooks/reset_camera_service.ipynb
my solution which didnt fix the problem but is a work around, is to restart the jetson nano.
here is another solution you can try, https://github.com/NVIDIA-AI-IOT/jetracer/issues/104

kumar132002, 2024-08-09:
I am stuck with the same issue and have tried reset_camera_service.ipynb ,Are there any other solutions