NVIDIA-AI-IOT/jetracer #134: cannot get the original image to work and make a bootable SD card, new image works perfectly

cyberluz, 2022-07-18:
I am not new to creating bootable SDs from an image file, and this one has me puzzled.

Downloading **jetcard_v0p0p0.zip** with WIndows 10 computer, using balenaEtcher I create the SD card, with a 64GB Sandisk SD card.  I first clear it with SD Formatter.  The process is flawless, no issues, but the completed card yields the nvidia splash screen that cycles on and off about every 10 seconds.

Doing the exact same process with **jetcard_nano-4gb-jp451.zip** I get a perfect bootable SD card!  No issues.

I tried Chrome vs Firefox, different SD card brands (Kingston vs. Sandisk), downloaded the zip via Mac ... nothing works.  The original image just doesn't work!  I tried balenaEtcher via URL, was warned about a missing partition table, but the same warning came up with the jp451 image, and still no success.

Am I crazy?  HELP!!!!!!!

Ford
Griffin Robotics

jaybdub, 2022-07-18:
Hi Ford,

Thanks for reaching out!  

This might be related to the image itself rather than how you are flashing the image.

I believe during the timeframe that the original image (v0p0p0) and the newer image (4g-jp451) were released, there was a change to the bootloader on Jetson Nano that causes older versions not to boot.

It's been a while since I've encountered this issue, so I'm hazy on the details, but something to be aware of.

Is it possible for you to accomplish your work on the latest JetCard image?  Or are there any challenges you're facing in doing so?

Best,
John

jaybdub, 2022-07-21:
Hi @cyberluz ,

Yes I believe when you use the later images, it updates the bootloader, making the older images not work.  

I am not sure if there is a way to revert this to use old images, but I can check.

If it's possible though, I would recommend moving towards a newer image and configuring JetRacer on top of that.  I'm happy to help with any issues you face doing this.

Best,
John

cyberluz, 2022-07-22:
We are working toward the new image, the reason we have encountered this.  The real issue is that some Nanos used to figure out what had happened came from Nanos used as Jetbots not Jetracers.  If they can't be reloaded/flashed I need to know that they are out of that use.  Not great as replacements are impossible given the supply shortages.

It would help if there is a way back.

Thank you,
Ford

cyberluz, 2022-08-25:
Is there a way back for these Nano boards so we can use them again as Jetbots not Jetracers?  I unknowingly switched some finding this issue.  Can I send them in to someone to be reverted for Jetbot use?

arickchan, 2022-11-25:
Hello @jaybdub 

New image work good in the training notebook but when I first run the road following notebook, error message shown as below:

---------------------------------------------------------------------------
OSError                                   Traceback (most recent call last)
<ipython-input-2-85080642f8f2> in <module>
----> 1 import torch
      2 import torchvision
      3 
      4 CATEGORIES = ['apex']
      5 

/usr/local/lib/python3.6/dist-packages/torch/__init__.py in <module>
    186     # See Note [Global dependencies]
    187     if USE_GLOBAL_DEPS:
--> 188         _load_global_deps()
    189     from torch._C import *
    190 

/usr/local/lib/python3.6/dist-packages/torch/__init__.py in _load_global_deps()
    139     lib_path = os.path.join(os.path.dirname(here), 'lib', lib_name)
    140 
--> 141     ctypes.CDLL(lib_path, mode=ctypes.RTLD_GLOBAL)
    142 
    143 

/usr/lib/python3.6/ctypes/__init__.py in __init__(self, name, mode, handle, use_errno, use_last_error)
    346 
    347         if handle is None:
--> 348             self._handle = _dlopen(self._name, mode)
    349         else:
    350             self._handle = handle

OSError: /usr/lib/aarch64-linux-gnu/libgomp.so.1: cannot allocate memory in static TLS block

Any idea to solve this problem? thanks
Rick Chan