NVIDIA-AI-IOT/jetracer #136: Testing examples, road_following

MaxKru, 2022-10-02:
Hello,
currently im trying out the given examples, until now everything worked out fine,
now when i try to go through the road following example i get the following error message,:
`
---------------------------------------------------------------------------
OSError                                   Traceback (most recent call last)
<ipython-input-2-85080642f8f2> in <module>
----> 1 import torch
      2 import torchvision
      3 
      4 CATEGORIES = ['apex']
      5 

/usr/local/lib/python3.6/dist-packages/torch/__init__.py in <module>
    186     # See Note [Global dependencies]
    187     if USE_GLOBAL_DEPS:
--> 188         _load_global_deps()
    189     from torch._C import *
    190 

/usr/local/lib/python3.6/dist-packages/torch/__init__.py in _load_global_deps()
    139     lib_path = os.path.join(os.path.dirname(here), 'lib', lib_name)
    140 
--> 141     ctypes.CDLL(lib_path, mode=ctypes.RTLD_GLOBAL)
    142 
    143 

/usr/lib/python3.6/ctypes/__init__.py in __init__(self, name, mode, handle, use_errno, use_last_error)
    346 
    347         if handle is None:
--> 348             self._handle = _dlopen(self._name, mode)
    349         else:
    350             self._handle = handle

OSError: /usr/lib/aarch64-linux-gnu/libgomp.so.1: cannot allocate memory in static TLS block
`
when trying to run following code:

`import torch
import torchvision

CATEGORIES = ['apex']

device = torch.device('cuda')
model = torchvision.models.resnet18(pretrained=False)
model.fc = torch.nn.Linear(512, 2 * len(CATEGORIES))
model = model.cuda().eval().half()
`

i already tried restarting, resaving the model i trained and this command
export LD_PRELOAD=/usr/lib/aarch64-linux-gnu/libgomp.so.1

how can i solve this error message?

ammarhaziqmh, 2022-11-08:
Have you manage to get a solution? I am having the same issue.

arickchan, 2022-11-24:
same issue here....

ammarhaziqmh, 2022-11-26:
> same issue here....

Try importing cv2 before import torch and torchvision as it helped me.

import cv2
import torch
import torchvision

CATEGORIES = ['apex']

device = torch.device('cuda')
model = torchvision.models.resnet18(pretrained=False)
model.fc = torch.nn.Linear(512, 2 * len(CATEGORIES))
model = model.cuda().eval().half()

didlawowo, 2024-11-17:
i have this error after adding cv2 
AttributeError: module 'torch' has no attribute '_six'