NVIDIA-AI-IOT/jetracer #149: About car models

rodsspt, 2024-01-10:
is this the only cars that will [Latrax version]
 [Tamiya version]. cause I have a car of my own I would like to know if it would work

coltonwestern, 2024-03-06:
I'm currently using a different car and it has been a hassle. I've managed to get everything to work up until the road following portion. It seems like a software issue though so I believe it's possible but you will most likely run into some compatibility issues. That being said, there are certain requirements that your hardware needs to meet to make sure it will work properly.

jaybdub, 2024-05-02:
@rodsspt,

It should be possible to adapt the road following to another platform, but it would take a bit of work.  You could implement a class similar to [this](https://github.com/NVIDIA-AI-IOT/jetracer/blob/master/jetracer/nvidia_racecar.py) to support your car.

That said, it has been a while since we've run through the examples, so they may not work out-of-the-box, given the updated software dependencies.

@coltonwestern do you mind sharing the issue you faced?

Best,
John