NVIDIA-AI-IOT/jetracer #150: Is this functional in 2024? I want my 11 year old try this over the summer break.

arjunkrishna, 2024-05-02:
Is this functional in 2024? I want my 11 year old son try this over the summer break.

I just want to check if this is still functional with parts available etc.

If you recommends any other uptodate project that would be nice too.

Thanks,
Arjun

jaybdub, 2024-05-02:
Hi @arjunkrishna ,

Thanks for reaching out!  Great to here you're interested in having your son explore robotics and AI.

It has been a while since we have touched this project.  Most of the hardware components should still work, but I can't say at this time how the out-of-the box experience would be.

There are other projects, like [DonkeyCar](http://docs.donkeycar.com/guide/build_hardware/#choosing-a-car), but to my knowledge, these still use the older Jetson Nano as well.

We are looking into updating these projects to support the latest Jetson Orin Nano.  And have an open [pull request](https://github.com/NVIDIA-AI-IOT/jetbot/pull/618) in JetBot to add instructions for a power pack that supports Jetson Orin Nano.  This may be another option.

Hope this helps, please let me know if you have any questions.  Also, if you have any more information on what you're looking for in a project to share with your son, we'd love to hear.

Best,
John

leonpano2006, 2024-10-02:
> Hi @arjunkrishna ,
>
>
>
> Thanks for reaching out!  Great to here you're interested in having your son explore robotics and AI.
>
>
>
> It has been a while since we have touched this project.  Most of the [...]

Is there a powerbank which is ideal to install for Xavier nx?
I use Xavier nx because I already have it